JarrettR/Stretch #1: Issue when stretching to SVG

KoolAidKid, 2020-09-21:
[image]
I believe this has something to do with missing plugins,but I could be wrong.

JarrettR, 2020-09-21:
Hey,

That's an issue with parsing the kicad PCB file. It just means that there's an edge-case in there that I haven't seen before.

If you can post that file, I can take a look at it at some point!

KoolAidKid, 2020-09-21:
Sure! what’s the best way to get you access ? just post it on github and
link?

JarrettR, 2020-09-21:
I think you can save it as pcb.kicad_pcb.txt and just attach it to this issue.

KoolAidKid, 2020-09-21:
Does this work?

[pcb.zip](https://github.com/JarrettR/Stretch/files/5258560/pcb.zip)

JarrettR, 2020-09-22:
That only has the .pro file, I just need the .kicad_pcb file

KoolAidKid, 2020-09-22:
[pcb (2).zip](https://github.com/JarrettR/Stretch/files/5259253/pcb.2.zip)
Oops!

JarrettR, 2020-09-23:
Fixed with https://github.com/JarrettR/Stretch/commit/ad1907ec0f83adfb4c3b62894ecca6ac964a133a

closed 2020-09-23 by JarrettR