JarrettR/Stretch #2: Windows depedency installation error

sujitth, 2020-10-09:
I was trying to install the dependency bs4 as mentioned in the instruction which gives an exception error as attached in the picture.

[image]

Can I install from the Cmd prompt(Python 3.8.5) like "pip install bs4"  instead of  "<Path>\KiCad\bin\python.exe -m pip install bs4 " ?
What is the difference?

JarrettR, 2020-10-09:
The latter command uses the portable version of Python that KiCad requires. It is 2.7 due to internal dependencies in KiCad, which is annoying, but I don't think that's the issue.

Can you try again, but with the command prompt opened with administrator privileges?

designer2k2, 2020-11-07:
got here the same initially, but with adminstrator privileges it worked!

closed 2020-11-07 by JarrettR