JarrettR/Stretch #3: documentation clarifications needed

mxa, 2020-11-30:
The installation instructions mention this step:

> Install dependencies using `D:\Programs\KiCad\bin\python.exe -m pip install bs4` (but substituting your own `KiCad\bin` path)

Is the assumption correct that this needs to be entered into the windows console with administrator privileges?
What is the equivalent instruction for other platforms, Linux in particular?
Is the extension designed and tested in kicad nightly or only in stable?

JarrettR, 2020-12-01:
- Yes, that is intended to be run on windows console with admin privileges
- For linux, it should work. With regard to the path, I'm not sure. I would assume that the Linux version of KiCad also comes with its own version of Python, but I do not know where that is located.
- Only tested on latest stable, 5.1.6. Happy to incorporate code that would allow compatibility with nightlies, as long as it does not break stable

mxa, 2020-12-01:
On Linux, KiCAD is using the systemwide python3 installation. libpython>3.8.2 is a dependency of the kicad package.
`sudo python3 -m pip install bs4` would thus be the equivalent way to get the bs4 package for python, or alternatively just `sudo apt install python3-bs4`if you want to use the system package manager over pip. Maybe you can add this info to the readme.

emosenkis, 2020-12-08:
Also, it would be nice if you could add the version(s) of KiCad that it's known to work well with.

JarrettR, 2020-12-29:
Done

closed 2020-12-29 by JarrettR