JarrettR/Stretch #4: Inclusion of DXF in tool

joshreve, 2021-01-07:
I am considering extending the tool to include DXF round trip translations along side SVG, as the formats and conversions are similar.  Would you be interested in having it added to your repo if I go down that path?  If I fork, both efforts may have less overall support.

JarrettR, 2021-01-07:
That's a really good question.

Currently, the bulk of this plugin is in only two Python files, with a total of about 2100 lines. I'd avoid doing too much work on the main branch, because I just keep adding edge-cases, and it's a mess.

I'm starting to chip away on the [refactor](https://github.com/JarrettR/Stretch/tree/refactor) branch. It'll be a while before it can even run, but the [via](https://github.com/JarrettR/Stretch/blob/refactor/stretch/via.py) file is roughly how all of the components will work. DXF, or any other format, should be straightforward to add.

JarrettR, 2021-12-18:
This took a little longer than I expected, but it would straightforward to add DXF, or any other format capability.

Straightforward, but still a lot of work.

joshreve, 2021-12-18:
I know how that goes.  I'm in the middle of a refactor on one of my repos.  I like this tool, but ended up going to a straight generation of the kicad file.  My use case was a bit different and found full programmatic generation more robust as I'm using geometries that are quite annoying to scale or stretch any other way.  I was originally planning to create files in a program with more dimensional control than kicad like QCAD/LibreCAD, but ended up creating a super simple geometry engine that just creates a kicad file directly.