JarrettR/Stretch #5: Unable to curve lines

emosenkis, 2021-01-07:
When I convert straight line segments to curves then reimport to KiCad, they continue to appear there as lines with the same endpoints instead of curves. Is there a specific way to create a curved board outline or trace?

JarrettR, 2021-01-07:
There are two ways, both of which have some downsides.

The way I prefer is to create many control points until the lines look smooth:

[image]

In the "edit path by node" mode in Inkscape, click on the lines that you want to smooth, and mash the "insert new nodes" button a few times. The downside here is that the line can't be easily adjust anymore.

The other method is to convert it to a KiCad curve.
In the XML editor (ctrl-shift-X), find the line's type:
[image]

Change it to gr_curve:
[image]

This allows you to even edit the curve as a bezier in KiCad, which is kind of neat. Unfortunately, you can only do this with lines / arcs / etc. _Segments_ are the copper traces, and if you're trying to convert them to curves, then they will no longer associate with PCB nets, and your ratsnests will show. If you get that manufactured, it's still work, electrically, though.

emosenkis, 2021-07-20:
I think this might be solved in the latest KiCad - maybe only in nightly. Would you consider adding support?

JarrettR, 2021-09-19:
I just skimmed the new changes in the code, and I don't believe anything is different.

Consider curves to be in the same category as arcs - You can't currently route tracks as arcs in KiCad. If that ever changes, then it's likely that curves will also be possible.

You can, however, create arcs using the "Place" menu, and assign it to the copper layer. It just won't associate with any particular net. Changing that to a curve in Inkscape's XML editor will work, with same caveat that it won't associate with any nets.