JarrettR/Stretch #7: Text severely messed up on export

holmesrichards, 2021-05-02:
Using:
Inkscape 1.0.2 (1.0.2+r75+1)
KiCad Version: 5.1.10-88a1d61d58~88~ubuntu20.04.1, release build
```
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-72-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

```
Linux Mint 20.1
Stretch latest

I tried Stretch export on a sandbox example and it badly messes up text on output. Font sizes are wrong, at least one text object is moved, and a bunch are gone entirely. And several %R references are unresolved.

KiCad original:
[image]

Stretch export:
[image]

Sonosus, 2021-05-03:
@analogoutput have you tried importing it back into KiCAD. I have the same problem but the text goes back to its original state when I reimport. From the readme: Fonts are iffy, and dimensions may be a little off. This imports properly back into the PCB if untouched, however.

holmesrichards, 2021-05-03:
I hadn't. Yes, it does import back into KiCad correctly. I do think this issue isn't well described by "fonts are iffy", though; it's a lot worse than that. But thanks for the clarification.

holmesrichards, 2021-05-03:
It's not just text. I created a polygon, an arc, and a circle in the F.Silk layer in PCBnew. On export to Inkscape the polygon and arc were there but the circle had disappeared. On re-import to PCBnew the circle was there.

In PCBnew:
[image]

In Inkscape:
[image]

JarrettR, 2021-05-03:
Proper font rendering in Inkscape is a challenging issue. I don't think it'll ever get fully solved in this tool, or similar. There are two issues at play: KiCad's font is unavailable in Inkscape, and KiCad also has independent width/height/stroke values that control how the text is rendered, which isn't really done in any other tool. Perhaps a workaround I could do is to instead draw the text as a vector line, and then vary the stroke width appropriately. It's worth adding to the feature list.

Not rendering designators is appropriate, I feel.  As we've already determined, the extent of text modifications you can reliably do in your vector software is repositioning and rotation, and maybe resizing. If you actually want to do some fun artsy stuff with it, you need to instead convert the text to polygons, so it's easy enough to rename the designator to whatever you like at that time. But current functionality will allow you to bring the PCB back into KiCad and not lose your reference information, which is one of the main goals of this project.

Looks like rotation inside modules is messed up in your original screenshots, too, which is something I do remember having issues with. I think I handled it better in the refactor branch (which is not usable yet, unfortunately).

Circle disappearing is another issue. It's possible that support for circles was never completed, I don't remember - It's slightly weird that KiCad treats arcs and circles differently.

JarrettR, 2022-07-18:
Won't fix, I feel this is adequately described in the readme, submit a PR with exposition if necessary

closed 2022-07-18 by JarrettR