JarrettR/Stretch #8: Oversized pads

Sonosus, 2021-05-03:
I'm not quite sure when this happened, but I noticed that the pads on one of my footprints are enlarged:
[image]
There is a workaround with right click -> update footprint:
[image].
It also shows up in Inkscape, but I imported/exported multiple times so it could've happened anytime.
[image]
I'm on lnkscape 1.0.2-2 (e86c870879, 2021-01-15)
KiCad :
```Application: KiCad
Version: (5.1.9)-1, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.71.0 OpenSSL/1.1.1g (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.73.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.71.0
    Compiler: GCC 10.2.0 with C++ ABI 1014
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON 
```
Windows 10
Latest Stretch.

JarrettR, 2021-05-03:
Looks like I haven't written in support for oval pads. Good to leave this issue open until it's eventually fixed

Sonosus, 2021-05-04:
Ok, thanks for responding.

JarrettR, 2021-12-18:
Refactor may have fixed this, support for unusual features is greatly expanded.

Reopen if necessary.

closed 2021-12-18 by JarrettR