JarrettR/Stretch #9: Changes not seen

holmesrichards, 2021-05-03:
I may just be misunderstanding the workflow or something, but the documentation does say:

"Make your modifications. Bend some lines, **draw some pictures,** rotate some footprints"

So I tried:

* Export to SVG
* In SVG, go into F.Silk layer
* Draw a circle (using the circle/ellipse tool)
* Change some text
* Save the file
* In KiCad hit down arrow, close and reopen PCBnew window

The PCB appears with the text changed, but no sign of the circle I drew.

holmesrichards, 2021-05-03:
Never mind, looks like my misunderstanding.

closed 2021-05-03 by holmesrichards

JarrettR, 2021-05-03:
Just to clarify where I think the issue was - If you draw a circle in Inkscape, it won't also have all of the metadata that is needed for bring back into PCBnew. You'll need to either add it in the XML editor by hand, or, preferably, draw it in PCBnew at the beginning and just use Inkscape to modify it, instead of creating it from scratch.