JarrettR/Stretch #13: Inkscape Beizer

NoNamedCat, 2022-01-11:
When i try to fatten the beizer curves in inkscape this error shows in the log:

Traceback (most recent call last):
  File "flatten.py", line 44, in <module>
    Flatten().run()
  File "/snap/inkscape/9256/share/inkscape/extensions/inkex/base.py", line 130, in run
    self.load_raw()
  File "/snap/inkscape/9256/share/inkscape/extensions/inkex/base.py", line 143, in load_raw
    document = self.load(self.file_io)
  File "/snap/inkscape/9256/share/inkscape/extensions/inkex/base.py", line 317, in load
    document = load_svg(stream)
  File "/snap/inkscape/9256/share/inkscape/extensions/inkex/elements/_base.py", line 92, in load_svg
    return etree.parse(stream, parser=SVG_PARSER)
  File "src/lxml/etree.pyx", line 3521, in lxml.etree.parse
  File "src/lxml/parser.pxi", line 1880, in lxml.etree._parseDocument
  File "src/lxml/parser.pxi", line 1900, in lxml.etree._parseFilelikeDocument
  File "src/lxml/parser.pxi", line 1795, in lxml.etree._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 1201, in lxml.etree._BaseParser._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 615, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 725, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 654, in lxml.etree._raiseParseError
  File "/tmp/ink_ext_XXXXXX.svgUCIJF1", line 2
lxml.etree.XMLSyntaxError: XML declaration allowed only at the start of the document, line 2, column 6

How can I fix it?

NoNamedCat, 2022-01-11:
In Inkscape on other SVG file I can make a beizer curve and flatten it with the extension normaly.

JarrettR, 2022-01-12:
I'm not sure I understand what you're trying to do. Is that an error that Inkscape is coming up with? I haven't had Inkscape be the source of any of my problems, yet.

NoNamedCat, 2022-01-13:
Yes. It's a problem with Ink-Scape when i try to use the plugin for flatten the beizer curves. It only shows this problem when i try to flatten the beizer curve in the file that drops Strech from kicad. On other SVG files I don't have this error.

JarrettR, 2022-01-13:
Can you post an example SVG?

NoNamedCat, 2022-01-13:
Yes. Tomorrow. Now i'm working. But when i get a little free time I will do it. I want to thank you for this awesome plugin!

closed 2022-07-18 by JarrettR