JarrettR/Stretch #14: Vias and Copper Zones

NoNamedCat, 2022-01-11:
I cant export the SVG file from kicad when vias or coper zones are in the PCB design.

When a via is in the design it shows this error:

Exception on python action plugin code

Traceback (most recent call last):

File "/home/user/.kicad/scripting/plugins/Stretch/Stretch/kiplug/-kicad.py" line 42, in Run
a.Run Plugin(pcb Filename, self.svg file name)

File "/home/user/.kicad/scripting/plugins/tretch/Stretch/kiplug/-
svg writer.py line 117 in Run Plugin
svg board.To_ SVG()

File /home/user/.kicad/scripting/plugins/stretch/stretch/kiplug/-
Doard.py line 254in To_SVG
tag-Beautifulsoup(item.To SVG) html.parser)

File "/home/user/.kicad/scripting/plugins/Stretch/Stretch/kiplug/
via py line 188, in To SVG
holefill#colourAssign( Via.lnner

TypeError Assign) missing 1 required positional argument: 'layername'

NoNamedCat, 2022-01-11:
When the copper zones are placed in the PCB design and i try to export the svg this came out:

Exception on python action plugin code
Traceback (most recent call last):
File "/home/user/.kicad/scripting/plugins/Stretch/Stretch/kiplug/-kicad.py", line 42, in Run
a.Run_Plugin(pcb_filename, self.svg_file_name)
File "/home/user/.kicad/scripting/plugins/Stretch/Stretch/kiplug/-svg_writer.py", line 117, in Run_Plugin
svg = board.To_SVG()
File "/home/user/.kicad/scripting/plugins/Stretch/Stretch/kiplug/-board.py", line 258, in To SVG
tag=BeautifulSoup(item.To_SVG(), 'html.parser")
File"/home/user/.kicad/scripting/plugins/Stretch/Stretch/kiplug/-
zone.py", line 278, in To_SVG
parameters +=";stroke:#' + Colour.Assign(self.layer)
TypeError: Assign() missing 1 required positional argument: 'layername'

JarrettR, 2022-01-12:
Thanks for the report, fixed

closed 2022-01-12 by JarrettR