JarrettR/Stretch #27: Stretch to SVG fails with exception

Papafelice, 2022-12-12:
Kicad V6.0.9, Linux
Exporting PCB to SVG fails with an exception in file text.py at line 237:
`if len(self.at) > 2:`
>`angle += float(self.at[2])`

because there is an additional parameter "unlocked":

(fp_text reference "S2" (at 0 4 unlocked) (layer "B.SilkS")

A line with an angle looks like this:

(fp_text reference "S2" (at 0 4 -45 unlocked) (layer "B.SilkS")

Changing the above lines to
`if len(self.at) > 3:`
>`angle += float(self.at[2])`

seems to fix the issue.

JarrettR, 2022-12-12:
Interesting. I'm unable to get KiCad to generate the `unlocked` parameter. Are you able to post a PCB file here, or test with the last commit I just pushed?

Papafelice, 2022-12-14:
Yes, that works.
To generate a footprint with "unlocked", just start the footprint editor, generate a new footprint with one pad and save:

`(footprint "Test" (version 20211014) (generator pcbnew)`
`  (layer "F.Cu")`
`  (tedit 0)`
`  (attr smd)`
`  (fp_text reference "REF**" (at 0 -0.5 unlocked) (layer "F.SilkS")`
`    (effects (font (size 1 1) (thickness 0.15)))`
`    (tstamp e2d35e77-ebb1-4709-b7d2-1d44ed9eb107)`
`  )`
`  (fp_text value "Test" (at 0 1 unlocked) (layer "F.Fab")`
`    (effects (font (size 1 1) (thickness 0.15)))`
`    (tstamp 66397689-3784-405c-97ff-0a0e6ce24f1f)`
`  )`
`  (pad "1" smd roundrect (at 0 -2.7) (size 2.286 1.524) (layers "F.Cu" "F.Paste" "F.Mask")` 
`(roundrect_rratio 0.25) (tstamp 104ba486-a7d4-4228-9eb7-950aee95ff94))`
`)`

closed 2022-12-15 by JarrettR