JarrettR/Stretch #30: `soupsieve` package is not installed

ONLYA, 2023-01-16:
When I installed and refresh this plugin, it shows "The soupsieve package is not installed. CSS selectors cannot be used."

I solved this issue by installing the required package to the KiCad python. However, is it possible to install all the required packages during the initialisation or installation stage?

ponzifex, 2024-03-14:
So, I have installed soupsieve like:
C:\Program Files\KiCad\8.0\bin\python.exe -m pip install soupsieve
Then, KiCad needs to be restarted to install Stretch.

JarrettR, 2024-03-15:
This has been "solved" by removing the internal beautifulsoup package and putting the responsibility on the user to install it.

`C:\Program Files\KiCad\8.0\bin\python.exe -m pip install beautifulsoup4` should do it

closed 2024-03-15 by JarrettR

ONLYA, 2024-03-19:
I mean, in perspective of user-friendliness, is it possible to automatically install the required package? For example, you can use the `pip` module to install it. The fallback of the `install` function to ensure it would work on all python versions. Even though it is not recommended, it is a good cross-platform solution. (I use Ubuntu instead of Windows)

JarrettR, 2024-03-19:
No, KiCad doesn't support automatically installing python module dependencies. I was trying to get around that by bundling BS4, but there are additional problems, due to how KiCad handles paths inside plugins.

Ideally, the KiCad PCM eventually supports dependency handling, and then this can be revisited.