JarrettR/Stretch #31: 'NoneType' object has no attribute 'next_element'

JarrettR, 2023-02-14:
This almost seems to be a bug in BeautifulSoup, although it hasn't seemed to crop up in KiCad 5 at all. It also only seems to happen when stretching up, right after stretching down.

It's generated from `base.svg.find('g', {'inkscape:label': layer}, recursive=True)` in `board.py`.
`layer` is `F.Cu` or anything else really.

The `base.svg` XML triggers the packaged BS in isolation, without using Stretch as a plugin:

```
<svg height="297mm" id="svg8290" inkscape:version="0.92.4 (5da689c313, 2019-01-14)" sodipodi:docname="base.svg" version="1.1" viewbox="0 0 210 297" width="210mm" xmlns="http://www.w3.org/2000/svg" xmlns:cc="http://creativecommons.org/ns#" xmlns:dc="http://purl.org/dc/elements/1.1/" xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape" xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#" xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd" xmlns:svg="http://www.w3.org/2000/svg">
<defs id="defs8284"></defs>
<sodipodi:namedview bordercolor="#666666" borderopacity="1.0" id="base" inkscape:current-layer="svg8290" inkscape:cx="-8.4507329" inkscape:cy="782.79942" inkscape:document-units="mm" inkscape:pageopacity="0.0" inkscape:pageshadow="2" inkscape:window-height="1137" inkscape:window-maximized="1" inkscape:window-width="1920" inkscape:window-x="-8" inkscape:window-y="32" inkscape:zoom="0.7" pagecolor="#000000" showgrid="false"></sodipodi:namedview>
<metadata id="metadata1">
<rdf:rdf>
<cc:work rdf:about="">
<dc:format>image/svg+xml</dc:format>
<dc:type rdf:resource="http://purl.org/dc/dcmitype/StillImage"></dc:type>
<dc:title></dc:title>
</cc:work>
</rdf:rdf>
</metadata>
<kicad></kicad><g attribs="user" id="layer58" inkscape:groupmode="layer" inkscape:label="User.9" number="58"></g><g attribs="user" id="layer57" inkscape:groupmode="layer" inkscape:label="User.8" number="57"></g><g attribs="user" id="layer56" inkscape:groupmode="layer" inkscape:label="User.7" number="56"></g><g attribs="user" id="layer55" inkscape:groupmode="layer" inkscape:label="User.6" number="55"></g><g attribs="user" id="layer54" inkscape:groupmode="layer" inkscape:label="User.5" number="54"></g><g attribs="user" id="layer53" inkscape:groupmode="layer" inkscape:label="User.4" number="53"></g><g attribs="user" id="layer52" inkscape:groupmode="layer" inkscape:label="User.3" number="52"></g><g attribs="user" id="layer51" inkscape:groupmode="layer" inkscape:label="User.2" number="51"></g><g attribs="user" id="layer50" inkscape:groupmode="layer" inkscape:label="User.1" number="50"></g><g attribs="user" id="layer49" inkscape:groupmode="layer" inkscape:label="F.Fab" number="49"></g><g attribs="user" id="layer48" inkscape:groupmode="layer" inkscape:label="B.Fab" number="48"></g><g attribs="user" id="layer47" inkscape:groupmode="layer" inkscape:label="F.CrtYd" number="47"></g><g attribs="user" id="layer46" inkscape:groupmode="layer" inkscape:label="B.CrtYd" number="46"></g><g attribs="user" id="layer45" inkscape:groupmode="layer" inkscape:label="Margin" number="45"></g><g attribs="user" id="layer44" inkscape:groupmode="layer" inkscape:label="Edge.Cuts" number="44"></g><g attribs="user" id="layer43" inkscape:groupmode="layer" inkscape:label="Eco2.User" number="43"></g><g attribs="user" id="layer42" inkscape:groupmode="layer" inkscape:label="Eco1.User" number="42"></g><g attribs="user" id="layer41" inkscape:groupmode="layer" inkscape:label="Cmts.User" number="41"></g><g attribs="user" id="layer40" inkscape:groupmode="layer" inkscape:label="Dwgs.User" number="40"></g><g attribs="user" id="layer39" inkscape:groupmode="layer" inkscape:label="F.Mask" number="39"></g><g attribs="user" id="layer38" inkscape:groupmode="layer" inkscape:label="B.Mask" number="38"></g><g attribs="user" id="layer37" inkscape:groupmode="layer" inkscape:label="F.SilkS" number="37"></g><g attribs="user" id="layer36" inkscape:groupmode="layer" inkscape:label="B.SilkS" number="36"></g><g attribs="user" id="layer35" inkscape:groupmode="layer" inkscape:label="F.Paste" number="35"></g><g attribs="user" id="layer34" inkscape:groupmode="layer" inkscape:label="B.Paste" number="34"></g><g attribs="user" id="layer33" inkscape:groupmode="layer" inkscape:label="F.Adhes" number="33"></g><g attribs="user" id="layer32" inkscape:groupmode="layer" inkscape:label="B.Adhes" number="32"></g><g attribs="signal" id="layer31" inkscape:groupmode="layer" inkscape:label="B.Cu" number="31"></g><g attribs="signal" id="layer0" inkscape:groupmode="layer" inkscape:label="F.Cu" number="0"></g><g inkscape:groupmode="layer" inkscape:label="Vias" type="layervia" user="True"></g><g inkscape:groupmode="layer" inkscape:label="Modules" type="module" user="True"></g><g inkscape:groupmode="layer" inkscape:label="Zones" type="layerzone" user="True"></g></svg>
```

More investigation needed.

mikhailmihalkov, 2023-02-19:
Does it work with kicad 7?

NoNamedCat, 2023-03-29:
I can confirm that bug. I replaced the contents of the folder BS with this source:
https://github.com/wention/BeautifulSoup4
And the plugin start working perfectly.
Can you update the source of BeautifulSoup in the Strech plugin please?

JarrettR, 2023-03-30:
Huh. That's bizarre, I don't believe that should actually work without relinking some of BS4's paths.

Regardless, I haven't been able to reproduce it on the BS4-dep-cleanup branch, so that might have fixed it. Unfortunately, I also haven't been able to reproduce the soupsieve bug in #30 reliably, so I'm sure that one isn't fixed yet.

closed 2024-03-15 by JarrettR