JarrettR/Stretch #36: Plugin doesn't load in KiCad 8.0 `No module named 'bs4'`

aocole, 2024-05-01:
Hello. Thanks for this plugin. I installed v1.3 via the plugin manager but the plugin doesn't load. The "plugin errors" dialog under preferences shows me the following:
```
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/aocole/Documents/KiCad/8.0/3rdparty/plugins/com_github_jarrettr_stretch/__init__.py", line 3, in <module>
    from .kiplug.svg_writer import SvgWrite
  File "/Users/aocole/Documents/KiCad/8.0/3rdparty/plugins/com_github_jarrettr_stretch/kiplug/svg_writer.py", line 4, in <module>
    from bs4 import BeautifulSoup
ModuleNotFoundError: No module named 'bs4'
```

JarrettR, 2024-05-02:
Thanks for the report.

To fix that, you'll need to BS4. So you'll have to `python -m pip install bs4` with whatever version of Python is being run by KiCad.

I'm not sure where that is on Mac - Windows it's at `C:\Program Files\KiCad\8.0\bin\python.exe` and Linux uses the system Python.

aocole, 2024-05-04:
Thanks - in my case on mac: 

```bash
cd /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/Current/bin
./pip3 install bs4
```

tisStrangeStuff, 2025-05-12:
I'm also having the same issue on windows, however I cant figure out how to install bs4, through the command prompt.

JarrettR, 2025-06-04:
You might try to open up the scripting console from the PCB editor and run:

```import sys
print(sys.path)```

And make sure that it's the same Python that you're installed BS4 onto. Also make sure to read #42 where you might need an older version of BS4 (but that's not yet your problem)

aerospace-venoms, 2026-08-18:
@ModelFKeyboard Hey, I ran into the same issue as you but on Kicad 10 on Debian. I ended up fixing the issue with some help from Claude. Basically your issue is your interpreter is mismatched from what the plugin actually uses. You ran pip install and got bs4 into your system's python 3.13 packages. However, KiCad is running its own bundled Python interpreter 3.11 and your new bs4 isn't there.

So what you need to do is find KiCad's version of python, and install bs4 for that. Do that by opening the Scripting Console in PCBNew, and run:
```
import sys
print(sys.version)
print(sys.path)
```

Now you know where KiCad's python is. You need to get your bs4 dependency somewhere that python will find it.

Godspeed, sir. Hopefully this response is helpful to over a year later.