JarrettR/Stretch #37: TypeError: can only concatenate str (not "list") to str

electronsandstuff, 2024-05-08:
Thanks for the plugin! I just installed it in kicad 8.0 from the plugin manager. After setting up bs4 (ie from #36) I tried to export one of my boards to SVG and get the following exception.
```
TypeError: can only concatenate str (not "list") to str
```
Here is a screenshot of the error message.
[image]

JarrettR, 2024-05-15:
Thanks!

I'll take a look when I have time. I'd suspect that changing `zone.py` line 287 from `self.keepout` to `str(self.keepout[0])` would work as a temporary stop-gap

juamonsan, 2024-07-24:
Hi, same issue here with last version of KiCAD and the plugin. Tried some workarounds in zone.py line 287 (str, .join, print, .format...) but same error appears every time.
If I have some time I will try more things and I will let you know.
Thanks in advance.

juamonsan, 2024-10-12:
Seems to work fine again after updating KiCAD to 8.0.5. SVG file is being exported correctly with the exact same project I had trouble with on 8.0.4.